HebiRobotics/hebi-matlab-examples #102: about motor torque controller

76328, 2019-11-18:
1.Is there any api for motor torque control?
2.Can I use the motor current and PWM to control motor torque for SEA output torque control algrothim design?
3.Dose the SEA support motor torque control like FOC?

daverolli, 2019-11-18:
> 1.Is there any api for motor torque control?
Yes.  The `effort` commands control the torque of the actuator.  This torque is measured and controlled directly on the output of the actuator using the sensed deflection of an internal spring.

> 2.Can I use the motor current and PWM to control motor torque for SEA output torque control algrothim design?
You can access the motor current in the actuator feedback, as well as command the PWM directly if desired.  So you can try closing the loop yourself for research.  However, actuators have high gear ratios, so it is unlikely that controlling current/torque of the motor directly would result in good control of output torque.

> 3.Dose the SEA support motor torque control like FOC?
The actuators do not currently support field-oriented control, or current-based control methods directly on the electric motor.  Torque control is provided by sensing directly on the output.

76328, 2019-11-18:
> > 1.Is there any api for motor torque control?
> 
> Yes. The `effort` commands control the torque of the actuator. This torque is measured and controlled directly on the output of the actuator using the sensed deflection of an internal spring.
> 
> An API example for torque control of a single actuator is here:
> https://github.com/HebiRobotics/hebi-matlab-examples/blob/master/basic/x-series_actuator/ex3c_command_effort.m
> 
> > 2.Can I use the motor current and PWM to control motor torque for SEA output torque control algrothim design?
> 
> You can access the motor current in the actuator feedback, as well as command the PWM directly if desired. So you can try closing the loop yourself for research. However, actuators have high gear ratios, so it is unlikely that controlling current/torque of the motor directly would result in good control of output torque.
> 
> > 3.Dose the SEA support motor torque control like FOC?
> 
> The actuators do not currently support field-oriented control, or current-based control methods directly on the electric motor. Torque control is provided by sensing directly on the output.

Wow,You means you use motor voltage (PWM) and a single PID loop to control the (output defelection)effort of SEA and achieve such high accuracy,it’s amazing! So Can I further ask an question,the PWM strategy is just like SVPWM technology?And there is no current based controller，you just control the output PWM which control the winding voltage，so the motor driver just like six step driver or sin wave driver not the FOC？

daverolli, 2019-11-18:
Correct, we have had good results by controlling the PWM to the motor (the winding voltage) and we do not use FOC.  

Hope that helps.  If you'd like discuss more we'd be happy to continue the discussion via email at support@hebirobotics.com.

closed 2019-11-18 by ennerf