HebiRobotics/hebi-matlab-examples #137: How is feedback volume 'deflection' calculated?

Tom0514, 2023-04-17:
Hello, everyone!

      I am doing a single joint test, the joint model is `X5-9`. But I ran into a doubt, `deflection` is not equal to the value after subtracting the equivalent motor position from the joint position. In theory, shouldn't they be the same?

Best regards,
Xu.

ennerf, 2023-04-17:
Deflection is calculated from two encoders that are located before and after the spring attached to the output, i.e., they are after the gear train. The motor position is the position of the motor itself, i.e., it is before the gear train.

`motor -> [motor_position] -> gear train -> [deflection as offset from position] -> spring -> [position]`

ennerf, 2023-04-17:
The torque can be calculated using [Hooke's law](https://www.jamesspring.com/news/spring-constant/), i.e., `effort = -stiffness * deflection`. Note that the stiffness portion is negative, which may explain your sign error.

That being said, the springs in the X- and R-series actuators have some hysteresis effects and newer models use non-linear springs, so the stiffness depends on multiple factors and is not straight forward to compute. Each spring gets calibrated independently before shipping.

You may want to take a look at the `Encoders` section in Scope's log-viewer for more details. In the log file below the `J2_shoulder`  (red) joint shows a non-linear spring while `J5_wrist2` (green) uses a more linear spring.

[image]

[image]

Tom0514, 2023-04-17:
Hello, thank you for your reply!

> Edit: I somehow missed your code comment. It looks reasonable to me, but I'll ask someone else to double-check.

This code is just to verify `deflection = theta_link - theta_motor/N`, where `theta_motor/N` perform zero correction.

> The torque can be calculated using [Hooke's law](https://www.jamesspring.com/news/spring-constant/), i.e., `effort = -stiffness * deflection`. Note that the stiffness portion is negative, which may explain the sign error.

I can get the corresponding torque value through Hooke's theorem, but it has positive and negative numbers throughout the constant positive speed operation (sorry, the expression in the previous reply is too absolute, and it has positive and negative numbers in the figure). From the point of view of the formula, it should not be positive or negative.

> You may want to take a look at the `Encoders` section in Scope's log-viewer for more details. In the log file below the `J2_shoulder` (red) joint shows a non-linear spring while `J5_wrist2` (green) uses a more linear spring.

I drew it and it's a linear spring. 
[image]

ennerf, 2023-04-17:
> This code is just to verify deflection = theta_link - theta_motor/N, where theta_motor/N perform zero correction.

That is not the case. The deflection comes from two dedicated absolute magnetic encoders and is independent of the motor and gear train. The motor-position is based on hall sensors and is not used in the effort computation at all. Between non-linearities in the encoders, gear backlash etc., I would expect the results to deviate a bit from each other.

> I can get the corresponding torque value through Hooke's theorem, but it has positive and negative numbers throughout the constant positive speed operation (sorry, the expression in the previous [...]

Could you send us a log file to support@hebirobotics.com? A subject line "hebi-matlab-examples#137" without any text would be enough.

Tom0514, 2023-04-17:
> That is not the case. The deflection comes from two dedicated absolute magnetic encoders and is independent of the motor and gear train. The motor-position is based on hall sensors and is not used in the effort computation at all. Between non-linearities in the encoders, gear backlash etc., some deviation between the two values is to be expected.

Thank you for answering my doubts. The calculation given by the dynamic model of the joint is too ideal. 

> Could you send us a log file to [support@hebirobotics.com](mailto:support@hebirobotics.com)? A subject line "hebi-matlab-examples#137" without any text would be enough.

OK, I'll send it right now.  Thank you for your answer, much appreciated!

ennerf, 2023-04-18:
The torque readings aren't great, but not out of whack for a 5 year old actuator. However, the commands are a bit weird.

You are using `DIRECT_PWM`, but something is making the command oscillate. Are you using some custom controller on top or maybe have safety limits set?

[image]

Tom0514, 2023-04-19:
Thank you so much! ennerf.

> You are using `DIRECT_PWM`, but something is making the command oscillate. Are you using some custom controller on top or maybe have safety limits set?

There are no other controllers on the top layer, I just control the PID to achieve constant speed movement. The safety limit parameters are set as follows:

``` matlab
safetyParams = group.getSafetyParams();
safetyParams.positionLimitStrategy = 3;        % damped spring
safetyParams.positionMinLimit = -3*pi;
safetyParams.positionMaxLimit = +3*pi;
```

Tom0514, 2023-04-19:
> What PID controller are you referring to? You are using DIRECT_PWM, which bypasses all built-in PID controllers.

Basically follow this example to write: [`ex1_custom_pid_1_controller.m`](https://github.com/HebiRobotics/hebi-matlab-examples/blob/master/advanced/ex1_custom_pid_1_controller.m)  The parameters used are `P` and `D`.

> Please try switching to Strategy 3 or 4 (see [documentation](http://docs.hebi.us/core_concepts.html#control-strategies)) and commanding velocity.

I saw in [`Choosing a Control Strategy`](https://docs.hebi.us/core_concepts.html#gain-tuning) that the `Direct PWM` strategy is suitable for system identification, so I used it. I also tried `strategy 4` and set the parameter of effort PID to 0, but the effect was worse than `strategy 1`. 
I didn't tune the parameters of the position and velocity PID in `strategy 4`, because I found that even when the position and effort were set to zero, the velocity and effort still fluctuated, and I don't know if it can achieve a constant and precise speed tracking. So I use `strategy 1` to track the position of the linear transformation, so as to realize the speed tracking indirectly.

Tom0514, 2023-04-19:
> Did you zero the commands or the gains? The position/torque controller should be disabled, e.g.,

Sorry, I didn't express myself clearly.  When the joint is not running, the speed and torque still fluctuate after setting the zero position in `Scope`. When I use `strategy 4`, I only send velocity commands: 

```matlab
cmd = CommandStruct();
cmd = CommandStruct();
```

closed 2023-04-21 by Tom0514