HebiRobotics/hebi-matlab-examples #139: Can a current controller be implemented with a direct PWM strategy?

Tom0514, 2023-05-10:
Hello, ennerf !

I'm testing the `MPC` algorithm in a joint, and my control input signal is the motor winding current, i.e. `windingCurrent`. When testing the joint, I scaled the value to the `[-1,1]` interval and sent it to the motor with a `direct PWM` strategy, but the actual effect was far from the simulation result. 

So, I want to track the current signal by cascading a bottom PID current controller. But when writing the current controller on the `direct PWM` strategy and applying it to the joint, the LED of the joint turns red and then makes a screeching sound. Is it possible to implement a current controller under `direct PWM` strategy? Thanks!

Best regards,
Xu.

ennerf, 2023-05-10:
It is generally possible to run a current loop on PWM (this is how FOC works in a nutshell), but it is generally timing critical and should be done in firmware. The actual equation that scales voltage to current in a motor should include the motor velocity and the winding resistance.

You are going to run into timing issues when doing it over the network, but you can maximize the feedback rate and hope for the best. As a first debugging step I'd recommend looking at logs and checking what your PWM commands look like. The red LED and sound probably indicate that your controller is unstable and oscillating.

Tom0514, 2023-05-11:
Thank you for your reply!

> You are going to run into timing issues when doing it over the network, but you can maximize the feedback rate and hope for the best. 

Doesn't changing the feedback frequency affect the sampling time of `MPC`?  My sampling time is `0.01s`, so I set the feedback frequency to `100Hz`.

> The red LED and sound probably indicate that your controller is unstable and oscillating.

In the early stage of `MPC`, the output current signal has a large oscillation. I think this may be the reason. I will try to add constraints to the control input rate.

For the joint `X5-9`, when setting the state variable of `MPC` and the upper and lower bounds of the control input, the parameters I set are:

```matlab
% The state variables correspond to motor position, motor speed, link position and link speed
lbx = -1*[pi, 14/9.55, pi, 14/9.55];    
ubx = +1*[pi, 14/9.55, pi, 14/9.55];
% control input: Motor winding current
lbu = -1.6; 
ubu = +1.6;
```

The current is set according to the peak current in the parameter table (`1.6A @ 36V`), the speed of the motor and link is obtained from the maximum speed (`14 RPM`), and the constraints of the position of the motor and connecting rod are self-selected (`+/- 4 rotations (8 x 360-degree total range)`). Is this reasonable? Thanks!

Tom0514, 2023-05-12:
> For the joint X5-9, when setting the upper and lower bounds of the state variable and the control input of MPC, the parameters I set are:

This can give `+1/-1` commands under the `direct PWM` strategy to observe the joint output, so as to obtain the relevant upper and lower limit values.

Tom0514, 2023-05-18:
Hello, ennerf!

I have designed a `PI` current controller with a feedback frequency of `1000Hz`, and when I am tracking a fixed value of 0.1A/0.2A, it is tracking ok. But today when I was trying to trace a sequence of control input signals from the `MPC` solution, it had a problem. 

Then when I try to track a higher fixed value of current under this `PI` parameter again, the feedback results are almost the same, and there is no tendency to track the target at all. 

Here is my feedback plot for tracking a fixed value of current using a `PI` controller (`P-0.2`, `I-50`): 
- [0.1A](https://github.com/Tom0514/issue/blob/main/0.1A.png)
- [0.2A](https://github.com/Tom0514/issue/blob/main/0.2A.png)
- [0.3A](https://github.com/Tom0514/issue/blob/main/0.3A.png)
- [0.4A](https://github.com/Tom0514/issue/blob/main/0.4A.png)
- [0.5A](https://github.com/Tom0514/issue/blob/main/0.5A.png)

The control input signals of the MPC and the corresponding simulation results (state trajectory and reference trajectory) are as follows:
- [the mpc control inputs](https://github.com/Tom0514/issue/blob/main/the%20control%20input.png)
- [state&reference trajectory](https://github.com/Tom0514/issue/blob/main/state%26reference%20trajectory.png)

Track the control input signal under the `PI` parameter:
- [track control input](https://github.com/Tom0514/issue/blob/main/mpc.png)

Is this a problem with the `PI` parameter? Why this happens, I have a hard time understanding. Here is my [code](https://github.com/Tom0514/issue/blob/main/hebi_current.m).

Hope you take a look! Thank you so much! 

Best regards,
Xu.

ennerf, 2023-05-19:
I tried running your code, and I think you are running into at least two issues:

1. You may be commanding something that is physically impossible. On an unloaded X8-3 going full tilt I get a maximum motorCurrent of 0.15 amps. I'd recommend doing a sanity check where you command max pwm and check what the output is for your system.

2. You are not capping the i term, so the controller is running into [integral windup](https://en.wikipedia.org/wiki/Integral_windup) and can't recover.

Tom0514, 2023-05-19:
Thank you for your reply!  I'm going to double-check the two points you raised.  Thanks again!!!

Tom0514, 2023-05-19:
> You may be commanding something that is physically impossible. On an unloaded X8-3 going full tilt I get a maximum motorCurrent of 0.15 amps.

Sorry, the output of my actual `MPC` is `windingCurrent`, I looked at the uploaded code and wrote `motorCurrent`. At that time, in order to verify whether it was a problem with tracking signals, I tested `motorCurrent`, but I looked at the log files, and it seemed that the tracking could not be achieved.

I'll check the problem again, I'm really sorry.

closed 2024-01-12 by Tom0514